pybop-team/PyBOP #12: Kalman / covariance propogation solver

BradyPlanden, 2023-07-18:
This issue is to decide the implementation language for the Kalman filter solver required for #3.

Options discussed:
Pure Python
JAX
PyBaMM's IDAKLU (C++)

Currently, JAX is the method being considered as it provides compiled performance as well as code modularity between CPU/TPU/GPU.

davidhowey, 2023-07-20:
Just to drop in, as an aside, that I do quite like the idea of the "halfway house" observer we discussed a few days ago, i.e., fixed feedback gain parameter K that is learnt as part of the optimisation. Any observer is better than none, and this might mean we can use existing solvers.

martinjrobins, 2023-11-29:
is this still in scope for pybop? I notice that the `BaseModel` class currently handles the timestepping, i.e. timestepping, or the way time is discretised, is considered part of the model, and data only comes into the problem. But the output of a KF is (for parameter estimation anyway) the likelihood, which is a cost function, and the timestepping is also handled by the KF (or any observer), and data comes into the KF. So its unclear where a KF would fit into the current design.

It would be great if this would be included still!

BradyPlanden, 2023-11-29:
Hi Martin,

Yes, I think everyone is still keen for this to be included; but I agree that it's not clear how it fits into the current design.  I believe the fixed-gain observer could be completed by inserting a fixed gain optimisation parameter into the RHS (although for which state variable is the question). 

For a true KF implementation (and without thinking too much about it), we might be able to formulate the KF as a cost function with methods to change the time step by changing the problem variables/methods. Is this something that you are interested in looking at?

martinjrobins, 2023-11-29:
sure, I can look at this. I'm not exactly sure how to implement a KF for a general pybamm model, but I'll look into it....

closed 2024-03-13 by BradyPlanden