pybop-team/PyBOP #30: Installation issue

NicolaCourtier, 2023-09-10:
### Python Version

>3.6

### Describe the bug

I tried to follow the instructions to install `homebrew`, `pyenv` and `pyenv-virtualenv` but unfortunately I encountered one of the Common build problems, specifically: https://github.com/pyenv/pyenv/wiki/Common-build-problems#error-the-python-ssl-extension-was-not-compiled-missing-the-openssl-lib

Although I was able to find a workaround from stack overflow, I suggest that we switch to the same installation instructions as PyBaMM for consistency: https://docs.pybamm.org/en/latest/source/user_guide/installation/GNU-linux.html#install-pybamm

### Steps to reproduce the behaviour

I encountered the issue on the first step of the installation instructions, when trying to create a pyenv-virtualenv. I am using WSL on Windows 10.

BradyPlanden, 2023-09-12:
Hi @NicolaCourtier, 

Can you confirm which pyenv you used? [pyenv-win](https://github.com/pyenv-win/pyenv-win) or [pyenv](https://github.com/pyenv/pyenv)?

The ongoing plan is to add multiple methods to readme's installation including the standard `virtual-env` method. If you can add that method to the readme that would be great.

NicolaCourtier, 2023-09-12:
Sounds good. To confirm, I was using pyenv in WSL.

closed 2023-09-15 by NicolaCourtier