pybop-team/PyBOP #98: Cost function returns an error when a simulation terminates early

NicolaCourtier, 2023-11-17:
If a PyBaMM simulation reaches a voltage constraint, the model can return a shorter solution array than expected, resulting in an error in the computation of a cost function such as RootMeanSquareError.

Optimisation problems are likely to run into such solutions when trialling different parameter values, but I think it should be taken to mean that the solution has a high cost rather than causing the optimisation to stop.

Suggested fixes:
1. Pad out any solutions that terminate early with the last output value to ensure that the prediction has the same length as the dataset.
2. Return a cost value of Infinity in the case that the prediction is shorter than the expected output.

I would opt for fix 2 for now, as I think it would give predictable behaviour and, in the case that all predictions are incompatible, it should return a final cost of Infinity.

BradyPlanden, 2023-11-20:
https://github.com/all-contributors, please add @NicolaCourtier to tests and examples.

BradyPlanden, 2023-11-20:
@all-contributors, please add @NicolaCourtier to tests and examples.

BradyPlanden, 2023-11-20:
@all-contributors
please add @NicolaCourtier for tests. 
please add @NicolaCourtier for examples.

BradyPlanden, 2023-11-20:
@all-contributors
please add @NicolaCourtier for test.

closed 2023-11-20 by BradyPlanden