pybop-team/PyBOP #190: Add another design cost

NicolaCourtier, 2024-02-09:
### Feature description

Add another design optimisation example, for example maximising the volumetric energy density.

### Motivation

The motivation is to provide an alternative to the existing GravimetricEnergyDensity for comparison purposes and applications where size is more important than weight.

### Possible implementation

Add VolumetricEnergyDensity and the required cell_volume function to the base electrochemical model.

BradyPlanden, 2024-02-09:
Great, that will be very helpful! 

I took a quick look at the open PR for this and was thinking about the best way to limit the number of new cost classes we need to add as we move forward. One way would be to merge `GravimetricEnergyDensity` and `VolumetricEnergyDensity` into something like `EnergyDensityCost` and provide an optional argument to select the appropriate geometric reference (i.e. `ref=volume` or `ref=mass`). This would reduce the number of lines we have to maintain and give us a clean interface to extend. Any thoughts?

NicolaCourtier, 2024-02-09:
Common base classes, like we have for models, would work for me. There will be other costs that require rebuilding of the model like these energy density costs do. So I think it makes sense to group design costs this way, rather than just pairing up the energy densities. Plus this way, we can keep the explicit names for each cost.

NicolaCourtier, 2024-02-09:
Shall I create a costs folder containing fitting and design cost base classes as part of this PR?

BradyPlanden, 2024-02-09:
Agreed, it's probably best to move the costs back into a directory with this change