pybop-team/PyBOP #222: Add electrode balancing example

NicolaCourtier, 2024-03-01:
### Feature description

Create an example workflow for fitting the electrode stoichiometry balance parameters from pseudo-OCV test data.

### Motivation

To demonstrate and benchmark a common battery parameterisation problem.

### Possible implementation

This could be implemented as an example script/notebook or as an extension of the FittingProblem class.

brosaplanella, 2024-03-14:
Similar to my comment in GITT: https://github.com/pybop-team/PyBOP/issues/223#issuecomment-1997057745. However, this one requires some extra work as our implementation do not conform to the PyBaMM model structure yet so it is a bit clunky (for ref https://github.com/paramm-team/pybamm-param/blob/develop/pbparam/optimisation_problems/OCP_balance.py). I know how the refactoring should go, just need to find some time to implement it.

BradyPlanden, 2024-03-21:
Assigned  - @DrSOKane

DrSOKane, 2024-03-21:
Would the half-cell OCP curves be given in terms of capacity or stoichiometry? Both work, but they're different optimization problems.

BradyPlanden, 2024-03-21:
Stoichiometry should allow for direct integration with PyBaMM, so that would be my preference.

brosaplanella, 2024-03-25:
In our implementation in pybamm-param (https://github.com/paramm-team/pybamm-param/blob/develop/examples/notebooks/ocp_balance.ipynb) we took the reference OCP to be vs stoichiometry and the measured one to be vs capacity. The idea was that OCP in the lab is measured vs capacity but, as Brady said, the models require it to be vs stoichiometry.

brosaplanella, 2024-08-22:
Regarding this example, the main issue is how to implement the model. I am not sure I fully understand the standalone model, should I just inherit from that class and define a new evaluate function? Or what is the recommended way to proceed?

PR #472 (2024-08-27): Add electrode balancing example
# Description

Added an example for electrode balancing. The main issue was defining the model, for now I have hacked it via an ECM but this means we can only balance half cells. In the future it would be interesting to define specific models for electrode balancing, which could support full cells too.

## Issue reference
Fixes #222 

## Review
Before you mark your PR as ready for review, please ensure that you've considered the following:
- Updated the [CHANGELOG.md](https://github.com/pybop-team/PyBOP/blob/develop/CHANGELOG.md) in reverse chronological order ( [... cut]