pybop-team/PyBOP #235: Improving cost function plotting?

davidhowey, 2024-03-13:
### Feature description

The cost landscape plots are super cool! I think it would be nice to make some minor changes to how they work to improve understandability. It would help users to interpret these plots if it was easier to see certain aspects. For example, it's not very obvious at the moment where the initial guess and the final (optimised) points actually are. Could we make these clearer, e.g., by circling them or similar, and/or adding some annotation text?

BradyPlanden, 2024-03-18:
Agreed, #198 changes the initial position from the red 'X' to a green circle, once #177 is merged we can update this with clearer annotations.

At the same time as the above, the method should be updated for vectorised inputs in the `model.simulate()` call. As the casadi solver can accept vectorised input in the form of a dict, this should speed up the plotting time considerably.

Reference: https://docs.pybamm.org/en/latest/source/api/solvers/base_solver.html#pybamm.BaseSolver.solve

BradyPlanden, 2024-07-05:
Things to consider adding:

- Scale for optimiser marker colouring (or more commentary in examples)
- Fix the axis titles when plotting \micro as .svg within the notebooks. See: https://github.com/pybop-team/PyBOP/blob/develop/examples/notebooks/spm_electrode_design.ipynb

PR #570 (2024-12-04): Improvement to landscape plots, MCMC summary, `x0` log entry
# Description

- Updates to MCMC summary class
- Update to contour and surface plots
- Adds x0 as optim.log entry

## Issue reference
Fixes #235 

## Review
Before you mark your PR as ready for review, please ensure that you've considered the following:
- Updated the [CHANGELOG.md](https://github.com/pybop-team/PyBOP/blob/develop/CHANGELOG.md) in reverse chronological order (newest at the top) with a concise description of the changes, including the PR number.
- Noted any breaking changes, including details on how it might impact exi [... cut]
closed 2024-12-04 by BradyPlanden