pybop-team/PyBOP #238: Allow multiple datasets & models at once

brosaplanella, 2024-03-14:
### Feature description

Allow the `FittingProblem` class to take lists of datasets and of models, which would streamline certain processes. The key idea is that when multiple models are present, they all use the same parameters (otherwise one can just write a for loop with N `FittingProblem`s).

Cases: 
- 1 model & 1 dataset: current case, just fit the model to the dataset.
- 1 model & N datasets: fit the model to minimise the cost function across datasets. A practical case for this would be when you run the same test on 3 different cells and want to fit the parameters.
- N models & N datasets: fit first model to first dataset, second to second and so on. Given that models are for specific operating conditions, an example of this would be if we wanted to fit a model to multiple C-rate experiments.
- N models & 1 dataset: I am still not convinced that this one would have any practical application, but it is the missing one so just listing it here.

### Motivation

Provided for each of the cases above.

### Possible implementation

Loop over and concatenate the error arrays before passing to cost function.

BradyPlanden, 2024-03-14:
Excellent thanks @brosaplanella, it looks like we were thinking along the same lines (#232). Happy to leave this one open as it has more background info.

To note, adding a weighting function to enable cost scaling across the datasets should be added at the same time as this functionality.

BradyPlanden, 2024-04-18:
Hi @MarkBlyth, just checking in to see how you are getting on with issue. If you are running into issues, or don't have the time to continue please let me know so that I can support or reassign.

MarkBlyth, 2024-04-19:
Hi Brady, I've been out the office for most of the past month so haven't had the chance to look at this yet. Still keen to work on it, and aiming to get started by the end of next week.

MarkBlyth, 2024-04-30:
Hi all, I've made a first pass, with code [here](https://github.com/MarkBlyth/PyBOP_develop/tree/%23238_multiple_datasets). Currently it will break the observer methods, hasn't been tested with multiple models, and has no unit tests, so it's very much work in progress.

The approach I've gone for is to make a DatasetCollection class, representing multiple datasets, and have the fitting_problem evaluate multiple models over multiple datasets, with the plotting and cost functions updated to accept multiple results at once. Given that this means making a large number of breaking changes to the code base, I was keen to get some feedback before I keep going with tests etc.

NicolaCourtier, 2024-04-30:
Sounds good, thanks @MarkBlyth! It would helpful at this stage to make a (draft) pull request so we can more easily see where you have made changes.

NicolaCourtier, 2024-06-14:
Hi @brosaplanella and @MarkBlyth,

I've made an example `MultiFittingProblem` class to tackle this open issue. It would be great if you could take a look at the draft PR #364. It's not a robust method yet, but I wanted to demonstrate the direction I think we should go in (similar to #329).

NicolaCourtier, 2025-10-29:
Before closing this issue as complete, let me state how to set up each type of optimisation problem using the new structure (from Oct 2025). Optimisation parameters with the same name are shared, so that one consistent set of parameters is obtained in each case.

1. "1 model & 1 dataset": 1 problem created with 1 simulator and 1 cost

2. "1 model & N datasets" (same experiment/input): 1 problem created with 1 simulator and 1 weighted cost combining N costs

3. "N models & N datasets" (different experiment/input): N problems created with N simulator-cost pairs, combined into 1 meta-problem

4. N models & 1 dataset: N problems created with N simulators and 1 cost, combined into 1 meta-problem