pybop-team/PyBOP #297: Automated issue generator for failed PyBaMM `develop` tests

BradyPlanden, 2024-04-18:
### Feature description

As discussed in #291, add the [report-log plugin ](https://github.com/pytest-dev/pytest-reportlog)for pytest with the [xarray issue creator](https://github.com/xarray-contrib/issue-from-pytest-log?tab=readme-ov-file).

An example implementation can be found here: https://github.com/keewis/reportlog-test

### Motivation

Add an automated issue for failed PyBaMM develop branch tests.

agriyakhetarpal, 2024-04-19:
The issue creator is working, though only partially – it can be seen in the PR on my fork above. I have submitted a patch to add the requested feature upstream, we should be good to go after that.

NicolaCourtier, 2026-02-06:
Hi @agriyakhetarpal, just checking in on this one - it sound useful! And since we have a couple of failing nightly dependency tests at the moment, might be a good time to test it out.

agriyakhetarpal, 2026-02-08:
Oh hi @NicolaCourtier – this was actually completed upstream a few months after my contribution back then, and it's surely useful. I never came back to this because I forgot 🤦🏻 I'll implement it now. Thanks!