pybop-team/PyBOP #320: [Bug]: Fix Integration tests post #304

BradyPlanden, 2024-05-10:
### Python Version

N/A

### Describe the bug

#304 has reduced the test suite completion time, but has resulted in a higher failure rate of the integration tests.

### Steps to reproduce the behaviour

See: https://github.com/pybop-team/PyBOP/actions/runs/9019436453/job/24782313287

NicolaCourtier, 2024-05-10:
I also saw this when I merged develop into #236, changing the scale in line 15 in tests/integration/test_parameterisations.py seems to help. I reduced it from 0.05 to 0.03 in that PR.

BradyPlanden, 2024-05-10:
Thanks @NicolaCourtier. I've opened a PR for this fix now. I was able to fix this issue without changing the range of potential parameter values. This was achieveable without extending the synthetic prediction, but the extension should help with identifiability of the fitting parameters in this case. 

It should be ready for review soon, I want to run the scheduled workflow on the branch for confirmation. On initial testing, this change appears to allow for tighter assertions on the parameterisations test, which would be great.

PR #321 (2024-05-17): Fixes #320
# Description

This PR updates the `test_parameterisation.py` integration tests. This is completed by hyper-parameter updates for the extended synthetic data and a change to the parameter priors to remove incorrect biasing to the MAP cost function.

In addition, it adds a prior resampling method to improve pybop.SciPyMinimize stability. This avoids failing test when `allow_infeasible_solutions == False` as x0 will be resampling when possible. To achieve this, the `pybop.Prior` classes have been refactored through the addition of a base class `pybop.BasePrior`.

## Issue reference
Fi [... cut]
commit e149cfbc (2024-05-17): Merge pull request #321 from pybop-team/320-bug-fix-integration-tests-post-304

Fixes #320
closed 2024-05-17 by BradyPlanden