pybop-team/PyBOP #326: Relax performance tests for`pybop.SciPyMinimize` and `pybop.GradientDescent`

BradyPlanden, 2024-05-14:
### Feature description

As per title.

### Motivation

As a follow up to #320, I'm suggesting that we greatly relax the performance tests for `pybop.SciPyMinimize` and `pybop.GradentDescent`. Both of these optimisers are flaky throughout the test suite, and don't seem to fair any better within the examples. We'll also need to update the documentation to ensure users understand that these optimisers are not actively tested for performance. 

Alternatively, if anyone wants to dive into the SciPyMinimize problems, they seem to be more stochastic than grad descent, which seems to require ongoing LR calibration.

I'd be interested to hear others' thoughts on this (@agriyakhetarpal, @NicolaCourtier, @martinjrobins, etc). Perhaps #255 will alievate some of the issues with Minimize by opening up better arg passing, but I'm not overly hopeful. Either way, this issue can wait until #255 is merged so that we can test.

agriyakhetarpal, 2024-05-14:
I probably won't have as much context as other people here, but generally, making improvements to the test suite towards making it reproducible would be a good way to start, i.e., by adding a random seed (any should work). The performance tests for these optimisers can be switched via environment variables or certain pytest flags to run on the self-hosted runner as well, given that CI runners will always be flaky to some extent.

BradyPlanden, 2024-06-07:
Closing this issue as it has been implemented in #306, #255, and #322

closed 2024-06-07 by BradyPlanden