pybop-team/PyBOP #341: [Bug]: Error on equivalent_circuit_identification.ipynb

victorventuri, 2024-06-03:
### Python Version

3.9.6

### Describe the bug

When attempting to run `examples/notebooks/equivalent_circuit_identification.ipynb`, an error occurs on the 5th cell, line 3: `values = model.predict(t_eval=t_eval)`

The error reads `AttributeError: 'ParameterSet' object has no attribute 'get'`

The traceback is below in the log output.

This is happening with PyBOP 24.3, and the underlying PyBaMM is 24.1.  I am still learning to use PyBOP and PyBaMM, so any help would be greatly appreciated. 

### Steps to reproduce the behaviour

This happened by just running the notebook. 

### Relevant log output

```shell
{
	"name": "AttributeError",
	"message": "'ParameterSet' object has no attribute 'get'",
	"stack": "---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[5], line 3
      1 sigma = 0.001
      2 t_eval = np.arange(0, 900, 3)
----> 3 values = model.predict(t_eval=t_eval)
      4 corrupt_values = values[\"Voltage [V]\"].data + np.random.normal(0, sigma, len(t_eval))
      6 # Form dataset

File ~/python_envs/pkgs_batt_test/lib/python3.9/site-packages/pybop/models/base_model.py:491, in BaseModel.predict(self, inputs, t_eval, parameter_set, experiment, init_soc)
    489 if self._unprocessed_model is not None:
    490     if experiment is None:
--> 491         return pybamm.Simulation(
    492             self._unprocessed_model,
    493             parameter_values=parameter_set,
    494         ).solve(t_eval=t_eval, initial_soc=init_soc)
    495     else:
    496         return pybamm.Simulation(
    497             self._unprocessed_model,
    498             experiment=experiment,
    499             parameter_values=parameter_set,
    500         ).solve(initial_soc=init_soc)

File ~/python_envs/pkgs_batt_test/lib/python3.9/site-packages/pybamm/simulation.py:91, in Simulation.__init__(self, model, experiment, geometry, parameter_values, submesh_types, var_pts, spatial_methods, solver, output_variables, C_rate)
     85         raise NotImplementedError(
     86             \"BasicDFNHalfCell is not compatible with experiment simulations.\"
     87         )
     89 if experiment is None:
     90     # Check to see if the current is provided as data (i.e. drive cycle)
---> 91     current = self._parameter_values.get(\"Current function [A]\")
     92     if isinstance(current, pybamm.Interpolant):
     93         self.operating_mode = \"drive cycle\"

AttributeError: 'ParameterSet' object has no attribute 'get'"
}
```

BradyPlanden, 2024-06-03:
Hi @victorventuri,

I suspect this is due to the changes we've done to the `ParameterSet` since `v24.3`. Can I confirm that you are using the example on the [main branch](https://github.com/pybop-team/PyBOP/blob/main/examples/notebooks/equivalent_circuit_identification.ipynb)?

victorventuri, 2024-06-03:
Hi, @BradyPlanden, thank you for the prompt reply! Yes, I am using the example from the main branch.

BradyPlanden, 2024-06-03:
Hi @victorventuri, that's interesting. I can't reproduce this bug on the `main` branch, but I can reproduce it on the `develop` branch. Would you mind sending me the operating system you are using so I can investigate further?

victorventuri, 2024-06-03:
Sorry @BradyPlanden, I should have explained better: the example notebook I am using is the one available in the `main` branch, but my local installation is the one from `pypi` (meaning, I don't know exactly which branch I am actually using, just whatever has been made available to be pip install-able)

victorventuri, 2024-06-03:
Oh, and I'm on macOS Sonoma 14.5

BradyPlanden, 2024-06-03:
Hi @victorventuri, ok, I'm also on macOS 14.5 so that's one less variable.

Our PyPI branch is `main`, which matches the released state of `v24.3` as we haven't released since. The example on main, `examples/notebooks/equivalent_circuit_identification.ipynb` works for me as expected when I've PyPI installed pybop (i.e. `python -m pip install pybop`) however, I am getting a nbformat installation requirement that seems to have been missed. 

When I try running the example on `develop` with the same installation, I get the error you've posted above. Can you copy the [example](https://github.com/pybop-team/PyBOP/blob/main/examples/notebooks/equivalent_circuit_identification.ipynb) from `main` and overwrite the local version you are using? Perhaps there has been cross-polination of the `main` branch locally.

victorventuri, 2024-06-03:
Hi @BradyPlanden, sorry for the delay in getting back to this, it's been a busy day. 

I started a brand new environment, and the first thing I did after creating it was `pip install git+https://github.com/pybop-team/PyBOP.git@main`. The problem persists for me.

BradyPlanden, 2024-06-04:
Hi @victorventuri, sorry to hear that. Can you upload the exact notebook you are using for this issue so I can debug it on my end. Thanks very much!

victorventuri, 2024-06-04:
Hi, @BradyPlanden. It appears github won't let me upload .ipynb files, so I've attached a .zip with the notebook inside. 
[ECM.zip](https://github.com/user-attachments/files/15558309/ECM.zip)

BradyPlanden, 2024-06-04:
Hi @victorventuri, thanks for sharing. The problem is in the third code-block:

```python
model = pybop.empirical.Thevenin(
    parameter_set=parameter_set, options={"number of rc elements": 2}
)
```
The parameter set API has changed since v24.3 (see: #246). I think you've got this example from the develop branch located [here](https://github.com/pybop-team/PyBOP/blob/develop/examples/notebooks/equivalent_circuit_identification.ipynb), whereas you need to use the one on the main branch, located [here](https://github.com/pybop-team/PyBOP/blob/main/examples/notebooks/equivalent_circuit_identification.ipynb). Note, these are two different files, one from v24.3 (main branch) and one from the most recent devlopments (upcoming v24.5).

To the point, here is the change you need to make:

```python
model = pybop.empirical.Thevenin(
    parameter_set=parameter_set.import_parameters(), options={"number of rc elements": 2}
)
```

As mentioned above, there will be an upcoming release in the next week or so. Until then, examples on the main branch should all work for you: https://github.com/pybop-team/PyBOP/tree/main. If you find any other bugs, please do raise more issues!

victorventuri, 2024-06-04:
Hi, @BradyPlanden. Thank you for your patience and for helping me with this; it's working! I apologize for not realizing the discrepancy between the branches. I'll keep my eyes peeled for the new release!

closed 2024-06-04 by victorventuri