pybop-team/PyBOP #358: Align the type of inputs passed from cost to problem and model

NicolaCourtier, 2024-06-12:
### Feature description

Align the `evaluate` and `simulate` functions of a problem/observer and model to accept only inputs of type `Inputs`. Update the `Parameters` functionality to return dictionaries rather than arrays.

### Motivation

To continue improvements enabled by the addition of the Parameters class in #322 and work towards enabling optimisation of mulitple costs/problems/models with different or overlapping parameter sets.

BradyPlanden, 2024-06-13:
Hi @NicolaCourtier, thanks for raising this. I'm keen to add the option to pass the `Inputs` type to the cost/problem/model, however is there a technical reason that this has to be the only type with the upcoming batch study changes?

In our current form, I very much like being able to do this: `cost([0.5,0.5])` as it's a very quick way to check the constructed objects, and interact cost/problem/model. I would like to keep the type interoperability for `list` and `np.array' if possible.

NicolaCourtier, 2024-06-13:
Hey @BradyPlanden, in the PR #352, I have kept the ability to pass values like this to the `call`, `evaluate` and `evaluateS1` functions of the cost (as the optimiser does).

Further down the chain (from cost to problem to model), I think we need to keep the parameter names associated with the values in order to prevent against cases where values are passed to the incorrect parameter. Having said this, I have also kept the ability to pass values directly to `quick_plot`. The quick way to pass values to a problem or model would be `parameters.as_dict(x)`. Please describe any other functionality you would like to see.

PR #359 (2024-07-04): Pass Inputs dictionary
# Description

Aligning Inputs between problem, observer and model.

## Issue reference
Fixes #358.

## Review
Before you mark your PR as ready for review, please ensure that you've considered the following:
- Updated the [CHANGELOG.md](https://github.com/pybop-team/PyBOP/blob/develop/CHANGELOG.md) in reverse chronological order (newest at the top) with a concise description of the changes, including the PR number.
- Noted any breaking changes, including details on how it might impact existing functionality.

## Type of change
- [ ] New Feature: A non-breaking change tha [... cut]
closed 2024-07-04 by NicolaCourtier