pybop-team/PyBOP #390: Least-squares regression

MarkBlyth, 2024-07-04:
### Feature description

Enable parameters to be computed using ordinary least squares, for linear fitting problems.

### Motivation

Fitting equivalent circuit models using purely optimisation-based methods turns out to not be particularly easy or robust, and needs careful tuning of optimisation methods, parameter bounds and priors, etc. However, for a given RC timescale, the optimal parameters of a Thevenin-style empirical model can be found directly, using ordinary least-squares. From experience, the fastest and most reliable way to fit these sorts of models is to (a) get a nonlinear optimiser to propose the RC time-constants; (b) for those RC time-constants, fit parameters using least squares; (c) use the resulting least-squares error as the cost for the nonlinear optimiser. Developing least-squares fitting in PyBOP would therefore make it a particularly useful tool for these sorts of parameterisation problems.

BradyPlanden, 2024-07-04:
Hi @MarkBlyth, 

Thanks for opening this issue! This is an interesting area to consider, I don't fit Thevenin style models that often, so it's interesting to hear your experience. In terms of OLS, we capture this estimator via the `GaussianLogLikelihood` classes, as OLS can be represented by maximum likelihood estimation (MLE) under a Gaussian distributed likelihood function [[1]](https://stats.stackexchange.com/questions/143705/maximum-likelihood-method-vs-least-squares-method)[[2]](https://stats.stackexchange.com/questions/12562/equivalence-between-least-squares-and-mle-in-gaussian-model). [This example](https://github.com/pybop-team/PyBOP/blob/develop/examples/scripts/spm_MLE.py) provides usage on the likelihood classes, but it would be great to more formally present this information more formally in a notebook, if you are interested in creating that while comparing Gaussian MLE with OLS :)

More generally, we could implement an OLS class that either constructs the `GaussLogLikelihood` class with additional features (?) or perhaps update the docstring and examples to better signpost this.

MarkBlyth, 2024-07-08:
>  In terms of OLS, we capture this estimator via the GaussianLogLikelihood classes

Yes that makes sense. `GaussianLogLikelihood` should cover everything needed for basic LSQ fitting, so I'll close this issue.

> it would be great to present this information more formally in a notebook

I've just raised issue #396 which should help in making use of `GaussianLogLikelihood` for Thevenin models. Once I've got it all built into a nice parameterisation method, I'd be happy to put it together into an example / demo to share.

closed 2024-07-08 by MarkBlyth