pybop-team/PyBOP #391: Add classical optimisation problems to testing suite & examples

BradyPlanden, 2024-07-05:
### Feature description

In addition to our current battery focused identification and optimisation tasks, we should add tests and examples with classical benchmark optimisation tasks. An initial list of potential functions is available [here](https://www.sfu.ca/~ssurjano/optimization.html)

### Motivation

This can help validate our workflows outside of battery focused tasks and provide benchmarks against other inference packages. Likewise, this could help the MCMC integration tests in #340 by having a higher performance forward model.

brosaplanella, 2024-11-21:
Do you mean throwing these functions to the optimisers or going through the PyBOP problem class? I think the former is more adequate. 

The other question I have is, should the examples be with the other examples? If we just use these functions for testing, we can define them in the tests folder. Otherwise, if we want to use them in the examples, we will have to define them inside PyBOP.

BradyPlanden, 2024-11-22:
> Do you mean throwing these functions to the optimisers or going through the PyBOP problem class? I think the former is more adequate.

The former would be my preferred method as well. I think an easy entry point is inheriting the BaseCost for these functions.

> The other question I have is, should the examples be with the other examples? If we just use these functions for testing, we can define them in the tests folder. Otherwise, if we want to use them in the examples, we will have to define them inside PyBOP.

That's a good question, I think it would be nice to offer these to users. In #481, I've created an `experimental` as discussed in the developers meeting, these classes could sit in this directory until we find the best location. Thoughts?

brosaplanella, 2024-11-22:
> That's a good question, I think it would be nice to offer these to users. In https://github.com/pybop-team/PyBOP/pull/481, I've created an experimental as discussed in the developers meeting, these classes could sit in this directory until we find the best location. Thoughts?

If they inherit from BaseCost we could incude them in asubdirectory in the costs folder

BradyPlanden, 2024-11-22:
Since they won't have a problem class, they won't be able to be used exactly like our other cost classes. This limitation could be captured in their docstrings, but the experimental directory does provide a clear warning to users for the beginning. Either works with me, as long as we are clear what the limitations are :) 

For reference the standalone cost class is a good starting place for this: https://github.com/pybop-team/PyBOP/blob/develop/examples/standalone/cost.py