pybop-team/PyBOP #396: Easy distinction between PyBOP parameters and PyBaMM parameters

MarkBlyth, 2024-07-08:
### Feature description

Okay slightly strange one here, but bare with me... It would be useful if it were possible to take the parameters proposed by PyBOP, and post-process them into the parameter set that PyBaMM sees. There's ways around this already, eg. use a custom / standalone model, which happens to be one of the usual PyBaMM models built with the post-processed parameters; or, subclassing a PyBOP model, and overriding the `simulate` function to make edits to the parameters before they go off to PyBaMM. Neither of these are particularly convenient though, so it would be good to have the structure in PyBOP to natively handle this parameter processing.

### Motivation

Three cases where I've found myself trying to do this, all in the context of fitting Thevenin models:

1. Functional parameters: PyBOP optimises over parameters (eg.) `r0_at_soc_1`, `r0_at_soc_0d9`, ..., and the postprocessor builds a `pybamm.Interpolant` from the pairs `(SOC=1, R0=r0_at_soc_1)`, `(SOC=0.9, R0=r0_at_soc_0d9)`, etc. for the PyBaMM model. PyBOP can't optimise over a `pybamm.Interpolant` and the PyBaMM model doesn't natively take R0s as a list of scalars, so the post-processor layer between PyBOP and PyBaMM gives a method to close this gap.
2. Least squares regression of ECMs: consider fitting a 1-RC model with a fixed time-constant, using the `GaussianLogLikelihood` method. PyBOP can be used to identify the RC resistance $R$, but to build a complete PyBaMM model, we also need to set the RC capacitance as $C=\tau/R$. This again needs some sort of intermediate step between the parameter(s) PyBOP suggests (in this case, $R$), and the parameters PyBaMM sees (here, $R$ and $C$).
3. Unsure how useful this will be in practice, but it would be interesting to see if fitting $1/R$ is more numerically stable than fitting $R$ directly. For a constrained optimisation where timescale $R*C _leq tau_{max}$, the search space for parameter $R$ can be particularly small (think the small strip close to the $y$ axis of a $y=1/x$ curve). In that case, optimising over conductance $1/R$ will have a much larger search space, and could therefore alleviate some of the difficulties the optimisers can face. Again this needs a post-processor / translation layer between the $1/R$ that PyBOP sees, and the $R$ that gets fed into the PyBaMM model.

### Possible implementation

Something like an optional callback function in the optimiser class. Takes a dict of named PyBOP parameters, and outputs a dict of named PyBaMM parameters.

martinjrobins, 2024-09-19:
I think what you are proposing is effectivly re-parameterising some base model to create a new model. I would suggest the best way to do this is to allow users to easily create and use a custom pybamm model with pybop. Taking your 2nd case, this might look something like:

```python
custom_model = pybamm.SPM()
custom_model.substitute(pybamm.Parameter("C"), pybamm.Parameter("tau") / pybamm.Parameter("R))
pybop_model = pybop.create_model(custom_model)
# ... use new_pybop_model to do the fitting
```
(note `pybamm.BaseModel.substitute` and `pybop.create_model` don't exist yet, but could easily be created)

NicolaCourtier, 2024-09-19:
Thanks Martin, this seems like an intuitive solution to the problem! Two questions:

1. How would the substitution method cope with cancellation? Take the example of transforming from (R, C) to (R ,tau=R*C). Do you think this implementation would substitute R*C for tau (desired behaviour) or R*tau/R (more literal)?

2. I think we should continue to support common PyBaMM parameter sets to promote data sharing. But it should be fairly straightforward for the user to convert between parameter sets, given that they are in control of the substitution. Would it be possible to generate parameter set conversion functions as well?

martinjrobins, 2024-09-19:
> 1. How would the substitution method cope with cancellation? Take the example of transforming from (R, C) to (R ,tau=R_C). Do you think this implementation would substitute R_C for tau (desired behaviour) or R*tau/R (more literal)?

There is a bunch of simplification code in pybamm, not sure it would work (i.e. cancel the R's) in this particular example. Something to look at and test I think.

> 2. I think we should continue to support common PyBaMM parameter sets to promote data sharing. But it should be fairly straightforward for the user to convert between parameter sets, given that they [...]

Yea that would be nice. You could do it but it would require storing the args of `subsitute` in the pybop model so you would have to rearrange the code above to:

```python
custom_model = pybamm.SPM()
pybop_model = pybop.create_model(custom_model)
pybop_model.substitute(pybamm.Parameter("C"), pybamm.Parameter("tau") / pybamm.Parameter("R"))
# ... use pybop_model to do the fitting
```

NicolaCourtier, 2024-09-20:
Sounds good! Thanks

NicolaCourtier, 2024-10-17:
Hi,

I've written an [example](https://github.com/pybop-team/PyBOP/tree/ecm_with_tau/examples/scripts/ecm_with_tau.py) (building on #533) that replaces the parameter `"C1 [F]"` with `"tau1 [s]"` before tackling the same parameter estimation problem as described in @MarkBlyth's example [ecm_tau_constraints.py](https://github.com/pybop-team/PyBOP/tree/ecm_with_tau/examples/scripts/ecm_tau_constraints.py).

I don't think the timescale is identifiable from this constant discharge data however, so to compare the numerical performance we would need to use more dynamic data (e.g. the pulse in [this notebook](https://nbviewer.org/github/pybop-team/PyBOP/blob/develop/examples/notebooks/equivalent_circuit_identification.ipynb)).

@MarkBlyth, I believe this example resolves case 2 (above) and that the method can be extended to the other cases as well, what do you think?

PR #557 (2024-11-19): Add ECM example with change of parameters
# Description

Add an example showing how to optimise over a change of parameters, e.g. (R, tau) pairs instead of (R, C) pairs.

## Issue reference
Fixes #396 (point 2).

## Review
Before you mark your PR as ready for review, please ensure that you've considered the following:
- Updated the [CHANGELOG.md](https://github.com/pybop-team/PyBOP/blob/develop/CHANGELOG.md) in reverse chronological order (newest at the top) with a concise description of the changes, including the PR number.
- Noted any breaking changes, including details on how it might impac [... cut]
closed 2024-11-19 by NicolaCourtier