pybop-team/PyBOP #414: Running comparing_cost_functions.ipynb throws error

Dibyendu-IITKGP, 2024-07-22:
### Python Version

Python 3.10.12

### Describe the bug

I was running the examples from the notebook folder. All of them run ok except one. comparing_cost_functions.ipynb is throwing error while calling rvs method.

### Steps to reproduce the behaviour

In the example folder go to notebook and then run comparing_cost_functions.ipynb.

NicolaCourtier, 2024-07-22:
Hi @Dibyendu-IITKGP, please refer to this discussion about notebooks: https://github.com/pybop-team/PyBOP/discussions/344
The `develop` version of the notebook should be run with the `develop` branch of PyBOP. Otherwise please use the `main` version.

BradyPlanden, 2024-07-22:
Interestingly, @Dibyendu-IITKGP has installed from develop. I'm reopening this to confirm if this is an OS specific issue, or if the pip install from develop is causing problems on Linux machines. I can confirm on MacOS that `install git+https://github.com/pybop-team/PyBOP.git@develop` in a new virtualenv allows the notebook in question to run correctly (after installing plotly).

@Dibyendu-IITKGP can you confirm you're jupyter instance is targetting the right virtualenv?

Dibyendu-IITKGP, 2024-07-22:
yes, it is.

Dibyendu-IITKGP, 2024-07-22:
Working now as per @BradyPlanden 's suggestion, modified %pip install git+https://github.com/pybop-team/PyBOP.git@develop --force-reinstall in the notebook file.

closed 2024-07-22 by Dibyendu-IITKGP