pybop-team/PyBOP #424: Compute initial concentrations from initial OCV

NicolaCourtier, 2024-07-25:
### Feature description

Currently, a `FittingProblem` can take an initial SOC value but it would be better to compute the initial concentrations from the open-circuit voltage (OCV) at the start of a measurement.

We should replace the `init_soc` option with the functionality to pass an `init_ocv`. When `init_ocv` is specified, the model should recalculate the initial concentrations before running the simulation.

### Motivation

Typically it is the initial OCV and not the initial SOC that can be obtained from voltage measurements.

MarkBlyth, 2024-07-31:
Is it possible to have the choice of specifying either an initial SOC or an initial OCV? Ie. adding an `initial_ocv` option, without removing the `initial_soc` kwarg, so that users can choose one or the other. (This might be what you're already planning, but just wanted to check and make sure!)

For fitting empirical models, it's more helpful to specify the initial SOC than the initial OCV, as the standard approach is to work with known SOCs, taken from a combination of coulometry and capacity-measuring experiments. Removing the option for specifying an `initial_soc` would therefore add an extra burden when working with these sorts of whole-cell data.

NicolaCourtier, 2024-07-31:
Short answer: yes!

For PyBaMM models, the initial SOC (for ECMs) or initial concentrations (for EChems) form part of the parameter set.
Therefore it will remain possible to define an initial state via the `parameter_set` of the model. In this case, an initial value does not need to be passed to the problem class.

For a `FittingProblem` where the initial SOC or concentrations are unknown, then these values can be determined from the initial OCV. In #425, this estimation can be performed automatically on each iteration of the optimisation by passing the `init_ocv` value to `FittingProblem`.

For a `DesignProblem` with an EChem model, the initial concentrations must be recomputed on each iteration for consistent evaluation of the design cost. For an ECM, the initial SOC is just part of the parameter set.

I hope that makes sense, I'll request a review for you to check when it's ready.

PR #425 (2024-08-07): Fix parameter set logic and allow FittingProblem to take init_ocv rather than init_soc
# Description

Remove the `init_soc` option from BaseProblem, keep the `init_soc` option for DesignProblem and replace it with an `init_ocv` option for FittingProblem and Observers. This change is to ensure consistency in problems with parameters that affect the calculation of the initial concentrations from SOC.

## Issue reference
Fixes #424

## Review
Before you mark your PR as ready for review, please ensure that you've considered the following:
- Updated the [CHANGELOG.md](https://github.com/pybop-team/ [... cut]
closed 2024-08-07 by NicolaCourtier