pybop-team/PyBOP #426: Fitting Problem with experimental data

L0tusC00kie, 2024-07-26:
Hi all,

I have not seen an example or documentation about implementing FittingProblem with experimental data, instead of the usage of synthetic data. Is this currently possible or eventually an upcoming feature of PyBOP? If it's the first case, I would be very grateful for a hint on how to do this, as I keep getting an error that DataSet must have _built_model, implying DataSet must have been a PyBOP Model itself with noise data? 

Thanks in advance and sorry if this question sounds stupid ...

NicolaCourtier, 2024-07-26:
Hi @L0tusC00kie, it's a bit hidden but there is one example notebook that fits experimental data. You can find it under examples/notebooks/LG_M50_ECM/1-single-pulse-circuit-model.ipynb.

Please take a look and see how you get on. You'll need to define a `pybop.Dataset`, a model and some `pybop.Parameters` to fit before passing these objects to a `pybop.FittingProblem`.

L0tusC00kie, 2024-07-26:
Hi Nicola,
thanks for the hint. I made a quick look, but not yet succeeded implementing in my own code.
I did define all the necessary inputs for FittingProblem.
[image]

time_measurement, voltage_measurement and current_measurement are all np.array type.

I'm curious that the experimental data may not conform PyBOP's convention for class Dataset?

MarkBlyth, 2024-07-29:
Hi @L0tusC00kie, I've been using PyBOP to fit empirical models from experimental data. I've shared the code I use [here](https://github.com/MarkBlyth/pybop_ecm), which you're welcome to use and adapt if you want. At present it's only really on GitHub for me to share it with students, so it's not necessarily a very clear piece of code, or at all documented, but I'm happy to talk further if any of it is useful to you.

You'll see the [datareaders.py](https://github.com/MarkBlyth/pybop_ecm/blob/main/datareaders.py) file has a set of functions that take experimental data from a battery cycler, and do things like loading the data in from a battery cycler output file, and extracting GITT pulses, etc. If you're using GITT pulses and have a Neware or BaSyTec battery cycler, they should work out the box for you. Otherwise they'll give you a good idea of how we can convert from a battery cycler output to the pybop Dataset objects.

I'd be happy to port some of these into PyBOP, if that's a route we want to go down.

Dibyendu-IITKGP, 2024-07-29:
Hi @L0tusC00kie I am using experimental data in one of the fitting problem. I use the following syntax (for me it's an .xlsx file):

import pandas as pd
file_loc = r'/home/Literatures/Samsung/test.xlsx'
df = pd.read_excel(file_loc, index_col=None, na_values=['NA'])
df = df.drop_duplicates(subset=['Time'], keep='first')
df = df[df['Time'].diff().fillna(0) > 0]
dataset = pybop.Dataset(
    {
        "Time [s]": df['Time'].to_numpy(),
        "Current function [A]": df['Current'].to_numpy(),
        "Voltage [V]": df['Voltage'].to_numpy(),
    }
)

Need to make sure the data is cleaned up with respect to 'time' column.

BradyPlanden, 2024-08-01:
Thanks @L0tusC00kie - closing now, feel free to reopen if you need more support. 

MarkBlyth - I think these could be useful as a compressed example notebook or in a separate repo under the @pybop-team for experimental fitting examples. Happy to chat further and work out a direction.

closed 2024-08-01 by BradyPlanden