pybop-team/PyBOP #433: [Bug]: Quick Plot not plotting.

MarkVeerasingam, 2024-08-04:
### Python Version

3.11.1

### Describe the bug

Hello all - i recently became of way of pybop, looks really cool!

When i ran the example ['equivalent_circuit_identification'](https://github.com/pybop-team/PyBOP/blob/develop/examples/notebooks/equivalent_circuit_identification.ipynb) I was able to run it fine until i reached the line "pybop.quick_plot(problem, problem_inputs=x, title="Optimised Comparison");" 

I was prompted to install kaleido, so i ran >pip install -U kaleido

after that when i ran the quick plot, i've been waiting over 5 mins for a plot to be plotted (at first this attempt this was 35+ mins).
I understand the computantitive resources invovled with simulation sw however being familiar with Pybamm and my machine specs im confident that there may be a bug here i am missing if anyone can help me out here I would be super thankful :)
[image]

### Steps to reproduce the behaviour

Installed pybop through cmd prompt as administrator.

when trying to install pybop regularly through cmd prompt with any administrator, greeted with:
ERROR: Could not install packages due to an OSError: [WinError 5] Access is denied: 'C:\\Users\\USER\\AppData\\Local\\Programs\\Python\\Python311\\Lib\\site-packages\\~asadi\\libcasadi.dll'
Consider using the --user option or check the permissions.

openened a vscode jupyter notebook

Followed  ['equivalent_circuit_identification'](https://github.com/pybop-team/PyBOP/blob/develop/examples/notebooks/equivalent_circuit_identification.ipynb)

BradyPlanden, 2024-08-12:
Hi @MarkVeerasingam,

Based on the error generated during installation, it appears that you don't have system privileges to install on this machine. I don't expect PyBOP to work in a half installed state. Were you able to fix this problem? If not, I would start there and confirm that the above plotting problem persists after a correct installation. We test plotting on Windows/Mac/Linux and haven't encountered the above, but if it is reproducible we will investigate. 

Thanks for raising an issue!

BradyPlanden, 2024-09-16:
Closing due to inactivity.

closed 2024-09-16 by BradyPlanden