pybop-team/PyBOP #443: [Bug]: nox doesn't run with conda

MarkBlyth, 2024-08-06:
### Python Version

conda python 3.11 / system python 3.8

### Describe the bug

noxfile.py won't operate within conda environment, and instead runs the system python installation.

Obvious answer is to not use conda, but my university-managed linux machine only offers python 3.8. IT services set me up with conda to work around this, but nox calls the system python (which is too old for PyBOPping), rather than the python 3.11 from conda.

### Steps to reproduce the behaviour

`conda create -n py311 python=3.11`
`conda activate py311`
`nox -s unit`

agriyakhetarpal, 2024-08-06:
Hi @MarkBlyth, if you install `nox` with `conda-forge` inside that environment and run it with your `conda` environment's Python, does that work? i.e., `conda install -c conda-forge nox` and then `python -m nox -s unit`?

closed 2024-08-06 by MarkBlyth

agriyakhetarpal, 2024-08-06:
Glad to see this worked for you! As an aside, the `conda` Python should be the one that gets prepended to the top of the PATH, at least inside an activated `conda` environment. I would recommend checking the PATH environment variable if it's the standard Python that's getting picked up.