pybop-team/PyBOP #478: Add applied current functionality to EIS methods

BradyPlanden, 2024-08-28:
### Feature description

First discussed in #417, this issue is to add applied current functionality to the EIS forward model.

### Motivation

The ability to incorporate applied current in the frequency domain impedance predictions enables non-stationary EIS methods for improvements in parameter inference.

rtimms, 2024-09-02:
I don't have the full context here, but is the forward simulation part of this something that can live in the EIS package? I'm conscious that integrating code directly in PyBOP and also developing pybamm-eis might lead to duplicated work or incompatibilities down the line.

BradyPlanden, 2024-09-09:
Hi @rtimms, I generally agree. This issue can probably be moved to pbeis, with PyBOP implementing the changes afterwards. Due to the way we build the forward model, I think we would need to refactor the underlying `BaseModel` to work directly with pbeis. Alternatively, we could look at building pbeis so that the API can be used by both PyBOP and casual users. At the moment I've implemented the pbeis method directly into PyBOP's BaseModel class, but in the future this could be rethought and updated.

rtimms, 2024-09-09:
We could also just put the `EISSimulation` straight into pybamm and get rid of pbeis. Not sure of the the way PyBOP uses the model and sim, but having the EIS in pybamm would simplify things and avoid conflicts.