pybop-team/PyBOP #494: [Bug]: Release v24.9.0 broken

BradyPlanden, 2024-09-11:
### Python Version

3.12

### Describe the bug

v24.9.0 is currently broken due to the optional `pybamm[cite]` dependency. The open issue on the PyBaMM repo [is here](https://github.com/pybamm-team/PyBaMM/issues/4432). 

We can either yank this release or release v24.9.1 with the optional dependency removed. This proceeds with a warning on import instead of an error. @NicolaCourtier  @martinjrobins @brosaplanella @agriyakhetarpal  Any preferences? I think patching to `v24.9.1` is probably mine.

### Steps to reproduce the behaviour

`pip install pybop`
`import pybop`

agriyakhetarpal, 2024-09-11:
Thanks for the ping, @BradyPlanden! Yanking the release sounds better to me, since PyBOP is unusable with PyBaMM v24.9.0 in this case. PyBOP v24.9.1 can be released afterwards when PyBaMM is ready.

BradyPlanden, 2024-09-11:
Fair shout - I've yanked it now.

closed 2024-09-18 by BradyPlanden