pybop-team/PyBOP #515: Add parser to import experimental data processed using PyProBE

tomjholland, 2024-09-20:
### Feature description

Add the capability to directly import a [PyProBE](https://github.com/ImperialCollegeLondon/PyProBE) `Result` object as a source of experimental data for PyBOP.

### Motivation

PyProBE provides a user-friendly interface to experimental data allowing fast filtering to the subset of data of interest for further analysis.

The [PyProBE README format](https://imperialcollegelondon.github.io/PyProBE/user_guide/writing_a_readme_file.html) would also provide value, closing the loop between the experimental and the PyBaMM Experiment that PyBOP uses for fitting.

### Possible implementation

1. Utility function to import PyProBE objects into the required format, following discussion that ensures common syntax between PyProBE and PyBOP. The PyProBE object can provide both the filtered raw data and corresponding PyBaMM step instructions.

2. Corresponding example demonstrating this integration.

BradyPlanden, 2024-09-24:
Hi Tom, thanks for opening this issue. PyProBE looks like an interesting package, we'd welcome a PR to flesh out the integration. Initially, I would recommend a notebook or script that can then be refactored to work within either the `pybop.Dataset` class or a utility method as you described. Nice job on the package!

brosaplanella, 2025-09-04:
This issue could probably be extended to read a `pybamm.Solution` object and create the dataset. Maybe it's easier to tackle the PyBaMM importer first, and then the PyProBE one.