pybop-team/PyBOP #518: [Bug]:  cannot import name 'greycomatrix' from 'skimage.feature'

JulWilh, 2024-09-26:
### Python Version

3.9.4

### Describe the bug

Hey all, 
I tried to import pybob with "import pybob" resulting in ImportError: cannot import name 'greycomatrix' from 'skimage.feature'. 

Package versions: 
pybob 0.26.2 
scikit-image 0.24.0

### Steps to reproduce the behaviour

Run "import pybob" with python

### Relevant log output

```shell
import pybob
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "[MyFolder]\pybob\__init__.py", line 5, in <module>
    from . import image_tools
  File "[MyFolder]\pybob\image_tools.py", line 16, in <module>
    from skimage.feature import greycomatrix, greycoprops
ImportError: cannot import name 'greycomatrix' from 'skimage.feature'
```

JulWilh, 2024-09-26:
I can do: from skimage.feature import graycomatrix, graycoprops

Is there a typo in the code or am I'm doing something wrong?

agriyakhetarpal, 2024-09-26:
Hi @JulWilh, this repository tracks the development for `pybop`, and not `pybob` (`p` vs `b`). This issue looks like it's better suited for opening under https://github.com/iamdonovan/pybob. :)

Side note: yes, I think `scikit-image` changed their instances of `grey` to `gray` across all their spellings some time down the line in their development.

closed 2024-09-26 by agriyakhetarpal

JulWilh, 2024-09-26:
Thanks. That was a stupid typo by myself.

agriyakhetarpal, 2024-09-26:
No worries at all, happens to the best of us!