pybop-team/PyBOP #532: Allow linked parameter updates for design optimisation

NicolaCourtier, 2024-10-10:
### Feature description

Allow optimisation parameters to update other model parameters via custom, user-defined functions.

### Motivation

To enable the optimisation of linked parameters. For example, allowing users to optimise the active material volume fraction, while updating the porosity so that their sum equals one.

### Possible implementation

This is easier to do for design optimisation due to the use of `model.predict`, which takes an updated `parameter_set` in `DesignProblem.evaluate`.

BradyPlanden, 2024-10-11:
This sounds great @NicolaCourtier, it would be great to implement this for both `FittingProblem` as well as `DesignProblem`, do you think that can be within scope of this issue? It could help reduce the number of infeasible solutions the optimisers experience during identification 😃

NicolaCourtier, 2024-10-11:
Thanks Brady, we are working on it! Hopefully it can be the same implementation for all problem types.

PR #533 (2024-10-17): Allow linked design parameters
# Description

Enables relations to be enforced between the optimisation parameters and other model parameters for the purpose of design optimisation. Adds an example showing how to link the porosity to the active material volume fraction.

## Issue reference
Fixes #532

## Review
Before you mark your PR as ready for review, please ensure that you've considered the following:
- Updated the [CHANGELOG.md](https://github.com/pybop-team/PyBOP/blob/develop/CHANGELOG.md) in reverse chronological order (newest at the top) with a concise description of the changes, incl [... cut]
closed 2024-10-17 by NicolaCourtier