pybop-team/PyBOP #642: [Bug]: Non-serializable PyBOP models can not be copied or parallelized

YannickNoelStephanKuhn, 2025-01-28:
### Python Version

3.11.0

### Describe the bug

PyBOP models can not be pickled, which makes them unusable with my current approach to integrate EP-BOLFI into PyBOP. I need to be able to `deepcopy` them; furthermore, them not being pickleable makes it impossible to parallelize their evaluation with `multiprocessing`.

It's definitely something inside PyBOP, as the original PyBaMM models serialize just fine.

Is that a known limitation, and if so, is it a necessary side-effect of some desired functionality?

### Steps to reproduce the behaviour

```
>>> from copy import deepcopy
>>> import pybamm
>>> import pybop
>>> deepcopy(pybamm.lithium_ion.DFN())
<pybamm.models.full_battery_models.lithium_ion.dfn.DFN object at [....]>
>>> deepcopy(pybop.lithium_ion.DFN())
```

### Relevant log output

```shell
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "[...]\Python311\Lib\copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "[...]\Python311\Lib\copy.py", line 271, in _reconstruct
    state = deepcopy(state, memo)
            ^^^^^^^^^^^^^^^^^^^^^
  File "[...]\Python311\Lib\copy.py", line 146, in deepcopy
    y = copier(x, memo)
        ^^^^^^^^^^^^^^^
  File "[...]\Python311\Lib\copy.py", line 231, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
                             ^^^^^^^^^^^^^^^^^^^^^
  File "[...]\Python311\Lib\copy.py", line 161, in deepcopy
    rv = reductor(4)
         ^^^^^^^^^^^
TypeError: cannot pickle 'module' object
```

BradyPlanden, 2025-01-28:
Hi Yannick,

This is somewhat known and does need to be fixed. At the moment, we are able to circumvent the problem with the Pints' based optimisers via their ParallelEvaluation class. But I believe this issue is showing itself in the SciPy optimisers (see: #590).

For the moment, you can try using the `new_copy` method within `BaseModel` to acquire a deep copy of the pybop model class. See [here](https://github.com/pybop-team/PyBOP/blob/02a2139fabd7d5737185729dbb88c1b5ab5122e8/pybop/models/base_model.py#L839) for more information. Let me know if this does the trick for the time being!

YannickNoelStephanKuhn, 2025-01-28:
Hi Brady, that would not solve the issue at hand, since I want to deepcopy an optimiser instance. Working around that may just be as much work as fixing the issue, so I'll have a cursory glance at it.

YannickNoelStephanKuhn, 2025-01-28:
A deep-dive into Python class handling later, it is done: https://github.com/pybop-team/PyBOP/pull/645

YannickNoelStephanKuhn, 2025-01-30:
There is one issue remaining: whenever a `problem` gets `evaluate`d, it consequently stops being pickleable. `deepcopy`ing it before evaluating "solves" that issue. So any parallel implementation might need to have a `deepcopy` step built-in.

It's the following attributes in problem.model._built_model that are not pickleable:
"y0S",
"rhs_eval",
"algebraic_eval",
"rhs_algebraic_eval",
"jac_rhs_eval",
"jac_rhs_action_eval",
"jacp_rhs_eval",
"jacp_algebraic_eval",
"jac_algebraic_action_eval",
"jacp_algebraic_eval",
"jac_rhs_algebraic_eval",
"jac_rhs_algebraic_action_eval",
"jacp_rhs_algebraic_eval",
"casadi_rhs",
"casadi_algebraic",

I didn't catch it initially since it does not occur when just running a PyBaMM model directly. Temporarily setting these attributes to `None` does not help either, because the "rhs_eval", "algebraic_eval", "casadi_rhs", "casadi_algebraic" are needed for all `evaluate`ions after the first one.

YannickNoelStephanKuhn, 2025-01-30:
https://github.com/pybop-team/PyBOP/issues/585 now contains an example of how to work around that parallelization issue.