pybop-team/PyBOP #699: restructuring pybop base model and building

martinjrobins, 2025-04-01:
### Feature description

A meta-issue to kick off the new pybop restructure.

We've got a draft of the classes involved in the restructure here: https://github.com/pybop-team/PyBOP2

New classes are:
- [ ] ProblemBuilder: Users create this first and use setters to set various aspects of a problem (e.g. costs, pybamm simulation, parameters). Finally a `build` function is called with creates a runner class (see below)
- [ ] BaseRunner / PyBammRunner: Created by the `ProblemBuilder` class, this is the class that is used by the optimisers. Represents the combination of a simulation and a cost function, takes in parameters and produces the evaluation of the cost function at those parameters.
- [ ] Costs: These are pybamm-specific cost functions that have a pybamm expression as the cost function, they are passed to the `ProblemBuilder` during the build process

martinjrobins, 2025-04-01:
Workflow is to do the restructure on a separate branch (`restructure`) and do individual feature branches off this, merging back into `restructure`. The CI doesn't have to pass to merge back to `restructure`

BradyPlanden, 2025-04-01:
Thanks for open this issue @martinjrobins!

A bit more information on the restructuring: 

- An initial example API is available here: https://github.com/pybop-team/PyBOP2/blob/main/example_usage.py
- I think we ended up agreeing on `Problem` instead of `Runner` where the `Problem` class/instance no longer relates to pybop's current `Problem` class but becomes an 'optimisation problem' which includes the underlying model, data, and cost functions. 

As an initial step, I'm happy to take on refactoring our current BaseModel into a builder pattern. I'm not confident this can be decoupled from creating a `Problem` class, but I we could start with a builder pattern that creates a model class, then we refactor to a `problem` class in follow on PR's. Thoughts?

martinjrobins, 2025-04-01:
yea, I think `Problem` works better. I'm happy if you want to do the `ProblemBuilder`. @NicolaCourtier did you want to have a go at the `PyBaMMProblem` (or I can?). Or I can have a go at a couple of the `Costs`? I figure they will all be interacting, but we can just all guess at the API from the other classes, then iterate until they fit together

NicolaCourtier, 2025-04-01:
Thanks @martinjrobins - if you could have a go at the `PyBaMMProblem` that would be great!

martinjrobins, 2025-08-11:
TODO Mon 11th:
- examples, notebooks
- check and increase coverage
- nicola's applications

Things to note:
- [multiprocessing](https://github.com/pybamm-team/PyBaMM/issues/5154)