pybop-team/PyBOP #739: Multiprocessing via pybamm solver

BradyPlanden, 2025-06-12:
### Feature description

This issue is to restructure the multiprocessing functionality to be at the solver level instead of at the sampler/optimiser level. At the solver level, the multiprocessing occurs in the C++ code, which provides a large speed up. 

The syntax for this looks like:

```python
import pybamm
import numpy as np
import time

# create the model
model = pybamm.lithium_ion.DFN()
param = pybamm.ParameterValues("Chen2020")

# change the current function to be an input parameter
param["Current function [A]"] = "[input]"

simulation = pybamm.Simulation(model, parameter_values=param)

# solve the model at the given time points, passing multiple current values as inputs
t_eval = np.linspace(0, 300, 500)
values = np.linspace(0.1, 1.0, 1000)
inputs = [{"Current function [A]": value} for value in values]
start_time = time.time()
sol = simulation.solve([t_eval[0], t_eval[-1]], t_interp = t_eval, inputs=inputs)
print(time.time() - start_time) # 7.83s
```
Compared to a single DFN solve at 0.223s (~28x improvement).

The items to consider:
- How does this interact with the non-population optimisers/samplers (i.e. conventional gradient-based methods)
- This will benefit from updating the default population size for the heuristic optimisers

Pybamm reference issue: https://github.com/pybamm-team/PyBaMM/pull/4449

martinjrobins, 2025-06-13:
Could allow a problem to take a list of params?

```python
class BaseProblem:
    # ... 
    def set_params(self, param_sets: list[np.ndarray]) -> None:
```

BradyPlanden, 2025-06-13:
Yep, I'm thinking the same thing. The `Parameter` class will need a corresponding update, but I think it will be a relatively straight forward pull. I've got an initial start on a local branch, so will self-assign this one.

BradyPlanden, 2025-08-05:
Closing with #742

PR #742 (2025-07-23): IDAKLU multiprocessing for pybamm pipeline
# Description

Adds IDAKLU multiprocessing for pybamm pipeline alongside `output_vars` for much faster optimisation.

### Benchmarks (M3 Pro Macbook):
[image]

The above is the optimiser runtime, which includes PyBOP overheads. The IDAKLU is set with end-point `t_eval` values and `t_interp` values matching the experimental observations. Additionally, the IDAKLU options are updated to reduce the number of retries on failed solutions. Further work in finding the best defaults should be investigated for optimisation processes that involve events or long  [... cut]
closed 2025-08-05 by BradyPlanden